UMONS-GFA/ardas #52: date no synced in ardas.py

bastinc, 2017-03-03:


kaufmanno, 2017-03-08:
Here is a code snippet to get a datetime from a pool of NTP servers and generate the ardas command to set the new date...

import ntplib
import datetime
from time import ctime
c = ntplib.NTPClient()
now = datetime.datetime.utcfromtimestamp(c.request('europe.pool.ntp.org').tx_time)
message = print('#SD %04d %02d %02d %02d %02d %02d' %(now.year, now.month, now.day, now.hour, now.minute, now.second))

bastinc, 2017-06-30:
implement a new cron task in python:
* Pause acquisition with #PA command #71
* to sync date (code snippet) 
* to  change file with #CF command

kaufmanno, 2017-07-04:
If the NTP server is not available, the raspberry pi clock should be set from the ardas RTC !

closed 2017-07-05 by kaufmanno