UMONS-GFA/ardas #68: Push annotations in InfluxDB

bastinc, 2017-03-20:


kaufmanno, 2018-01-20:
It would be a good practice to ask for a 'restart' tagged annotation at each restart (but do not block automatic restart with a default value such as 'reboot' for tag 'restart') and to allow annotations trough a master connection (telnet)

for inserting annotations see also https://maxchadwick.xyz/blog/grafana-influxdb-annotations

kaufmanno, 2018-01-20:
It turns out that there are two types of annotations:
- the grafana native (including the interactive annotations created with a ctrl + left mouse button) 
- the annotations based on another data source such as InfluxDB and retrieved via a query

closed 2018-01-22 by kaufmanno

kaufmanno, 2018-01-23:
Influxdb annotations implemented in v1.1.3.1